fga-eps-mds/2019.1-unbrake #4: Linguagem oficial do repositório

Bumbleblo, 2019-03-22:
Gostaria de abrir uma discussão pública sobre a linguagem oficial que vamos tomar na documentação e código deste projeto.

Bumbleblo, 2019-03-22:
Acredito que a melhor língua para o projeto seria a língua inglesa. Praticamente quem trabalha com tecnologia (pessoas que poderiam ajudar com o projeto) tem algum nível de conversação em inglês e também um [percentual considerável do planeta consegue se comunicar em inglês](https://pt.wikipedia.org/wiki/Lista_de_l%C3%ADnguas_por_total_de_falantes).

icaropires, 2019-03-22:
Na mesma linha do @Bumbleblo, sugiro que todas as questões relacionadas a desenvolvimento sejam em inglês, abrindo assim a possibilidade de contribuições e consultas a toda a parte da comunidade do Github conhecedora de inglês, que é bem maior do que a de português.

Já para documentações não diretamentamente relacionadas ao desenvolvimento, acredito que não trará grandes prejuízos serem elaboradas em português (pelo menos incialmente).

Bumbleblo, 2019-03-22:
Gostaria da opinião da equipe atual do projeto: @icaropires @joaorobson @ViniciusBernardo @VictorLeviPeixoto @GabrielTiveron  [lleticiaarj](https://github.com/leticiaarj) @tmcstiago @medeiroslucas

VictorLeviPeixoto, 2019-03-22:
Eu concordo com sua opinião @Bumbleblo, também gostaria que a linguagem oficial do projeto seja a inglesa.

ViniciusBernardo, 2019-03-22:
Eu concordo com a opinião do @icaropires de que apenas questões relacionadas ao desenvolvimento devem ser em inglês.

icaropires, 2019-03-22:
[image]
@joaorobson I guess it's decided.

Bumbleblo, 2019-03-23:
Yep, I will link that issue in #6 and write that in contributing guidelines.

Bumbleblo, 2019-03-23:
Linked! I'll close that issue now.

Bumbleblo, 2019-03-27:
As mentoras[Carla Rochar](https://github.com/RochaCarla) e [Bruna Moreira](https://github.com/brunanayara) chegaram a conclusão que o uso da língua portuguesa seria melhor para o acompanhamento do projeto. Como ambas são mentoras do projeto vamos seguir essa sugestão.

closed 2019-03-27 by Bumbleblo