fga-eps-mds/2019.1-unbrake #6: Write a contributing guide for project

Bumbleblo, 2019-03-22:
## Description

Write a initial contributing guide for our project. That document will contain some sections about how contributing, branch creation rules, issues proposals.. etc.

## Tasks

Write that sections in a initial version

- [ ] What is the purpose of that project
- [ ] How to find information about that code
- [ ] How to contribute (branch political, issue creation, etc)
- [ ] Style guide

## Additional Information

Most part of the open source projects in github don't have a contributing guide, so I do not have a good contributing guidelines to inspire. I'll proposal some organization similar with [Atom Guidelines](https://github.com/atom/atom/blob/master/CONTRIBUTING.md#contributing-to-atom)

Bumbleblo, 2019-03-23:
How it was decided in #4 the repository language will be English for github and portuguese to documents.

icaropires, 2019-03-23:
We (@Bumbleblo, @joaorobson and @ViniciusBernardo ) talked on slack and we think that #n-name-of-issue is a good pattern for branch names and @joaorobson suggested to abbreviate names when possible, I agree with him.

icaropires, 2019-04-01:
@Bumbleblo, vc se esqueceu de marcar quais tasks foram concluídas.

icaropires, 2019-04-01:
Relacionada a #34

closed 2019-04-01 by icaropires