fga-eps-mds/2019.1-unbrake #13: Setup initial CI

icaropires, 2019-03-23:
## Description

Setup initial CI to avoid messy code being sent to devel branch.

## Tasks

- [ ] Choose a CI tool and integrate-it
- [ ] Run tests
- [ ] Run linter
- [ ] Make build fail if any of past steps fail

## Additional Information

Bumbleblo, 2019-03-23:
That CI pipeline have code style step ?

icaropires, 2019-03-24:
@Bumbleblo Yes, it's included in linter step.

icaropires, 2019-03-24:
I'd like to use gitlab Ci for this project, but according to gitlab in [this page](https://about.gitlab.com/solutions/github/), it's not totally free:

[image]

icaropires, 2019-03-24:
I don't want to prepare physical machines to run CI because we'd need a physical machine, public IP, etc. So I won't use Jenkins.

icaropires, 2019-03-24:
I'll use Travis CI which offers the infrastructure, support for the languages we'll use and docker. Besides, it's free for open source projects with no expiration time.

icaropires, 2019-03-24:
Maybe we'll need to test for Windows, which is in early stage for Travis. If we need to run the code of this repository on Windows, and Travis environment don't be enough, I can change the CI environment. But for now Travis CI will do.

icaropires, 2019-03-29:
Issues and Pull requests will now be in english. See #4.

icaropires, 2019-03-30:
Uma opção para checagem de estilo poderia ser o [PrettierCI](https://www.prettierci.com/) mas não será realmente útil para o nosso projeto, uma vez que toda a formatação é feita por git hooks no ambiente de todos usuários que tenham executado chamado o entrypoint do docker pelo menos uma vez (ver solução da issue #18 )

Além disso o script de checagem pode ser facilmente chamado na integração contínua original.

commit c1bfc2b4 (2019-03-30): Add initial CI setup and needed node scripts

Resolves #13. Scripts will also help on running all checks locally
commit 114b6d1c (2019-03-30): Add initial CI setup and needed node scripts

Resolves #13. Scripts will also help on running all checks locally
commit 5236328e (2019-03-30): Add initial CI setup and needed node scripts

Resolves #13. Scripts will also help on running all checks locally
commit abe40495 (2019-03-30): Add initial CI setup and needed node scripts

Resolves #13. Scripts will also help on running all checks locally
closed 2019-03-30 by joaorobson