fga-eps-mds/2019.1-unbrake #7: Create Sidebar Menu

ViniciusBernardo, 2019-03-23:
## Description

Implement the main Sidebar menu of the application creating the base architecture for react-dom's router selection. That base architecture must have the four initial modules (configuration, calibration, test and analysis).

## Tasks

- [ ] Create link buttons for each of the four modules 
- [ ] Create the routes for each of the four modules
- [ ] Link each button of the first task with its respective route
- [ ] Create sidebar and router selector in decoupled components
- [ ] Use the material-ui components

## Additional Information

VictorLeviPeixoto, 2019-03-26:
Existe algum template para ser seguido na construção da sidebar?

ViniciusBernardo, 2019-03-26:
Yes. You can use the dashboard example at the material ui documentation. Here is the link https://material-ui.com/getting-started/page-layout-examples/dashboard/

VictorLeviPeixoto, 2019-03-26:
Can i use react-sidenav?

joaorobson, 2019-03-26:
Como um substituto do Menu do Material-UI?

closed 2019-04-03 by joaorobson