fga-eps-mds/2019.1-unbrake #36: Criar autenticação de usuário

joaorobson, 2019-03-31:
## Descrição

Criar autenticação de usuário utilizando o Django e provê-la por meio da página de login criada na issue #8.

## Tarefas

- [ ] Criar classe para usuário utilizando a configuração básica provida pelo [django admin](https://docs.djangoproject.com/en/2.1/topics/auth/).
- [ ] Utilizar o [JWT](https://getblimp.github.io/django-rest-framework-jwt/) para geração de tokens.
- [ ] Assegurar que senha está sendo salva encriptada.
- [ ] Realizar a integração da autenticação com a página de login.
- [ ] Criar um mecanismo que seja seguro para prover o token para o frontend. 

## Informação adicional

A maneira mais fácil de prover o token para o React é utilizando localStorage ou sessionStorage (algo como uma "memória local" acessível pelo JS). Porém, essas formas não são recomendáveis pela possibilidade de serem suscetíveis à ataques, como explicado [aqui](https://dev.to/rdegges/please-stop-using-local-storage-1i04). Nesse mesmo link é explicado como realizar essa tarefa de maneira segura utilizando cookies.

medeiroslucas, 2019-03-31:
Qual parte dessa issue pertence ao front-end? @joaorobson

joaorobson, 2019-04-01:
As duas útlimas tarefas, @medeiroslucas.

VictorLeviPeixoto, 2019-04-01:
A issue em si é para criar somente o login, sem signup?

RochaCarla, 2019-04-08:
critérios de aceita[ção!

icaropires, 2019-04-08:
> 
> 
> critérios de aceita[ção!

@RochaCarla, essa issue ainda será evoluída quando eu souber melhor o que vou fazer.

Mas estamos mapeando os critérios de aceitação também como tasks. O revisor do PR avaliará se todas as tasks realmente foram cumpridas. Tem problema fazer dessa forma?

GabrielTiveron, 2019-04-09:
É viável a utilização do pacote django.contrib.auth.models para a criação de um usuário?

joaorobson, 2019-04-09:
Sim, ele já vem incorporado no Django, certo?

GabrielTiveron, 2019-04-09:
Sim

joaorobson, 2019-04-09:
Então está ok usá-lo

Bumbleblo, 2019-04-12:
@GabrielTiveron conseguiu usar o jwt para gerar os tokens ?

GabrielTiveron, 2019-04-12:
O graphql tem uma função pra isso, usei ela @Bumbleblo

icaropires, 2019-04-12:
> O graphql tem uma função pra isso, usei ela @Bumbleblo

@GabrielTiveron , então acho que seria o caso de marcar a task como concluída na descrição da issue.

closed 2019-04-14 by joaorobson