fga-eps-mds/2019.1-unbrake #39: Criar formulário para calibração de temperatura

joaorobson, 2019-03-31:
## Descrição

Criar formulário para calibrar informações relativas aos sensores de temperatura do simulador.

## Tarefas

- [ ] Utilizar [redux-form-material-ui](https://github.com/erikras/redux-form-material-ui/tree/5.0) para criar os campos editáveis.
- [ ] Integrar o gráfico criado na issue #11 na página criada.
- [ ] Utilizar Material-UI para criar o restante dos componentes.
- [ ] Relacionar ações dos botões "Plota temperatura 1" e "Plota temperatura 2" com exibição das linhas no gráfico.

## Informação adicional

Abaixo, página correspondente no software utilizado atualmente: 

[image]

A parte superior onde existem abas de navegação e os botões da parte inferior não necessitam ser implementados na realização dessa issue.

Bumbleblo, 2019-04-03:
Quem estiver com essa issue por favor se assine conforme está escrito no [guida de contribuição](https://github.com/fga-eps-mds/2019.1-unbrake/blob/devel/CONTRIBUTING.md)

Bumbleblo, 2019-04-12:
O merge do pull request já foi feito, porque a issue ainda não está na pipe close ? Existe alguma justificativa ?

closed 2019-04-12 by joaorobson