fga-eps-mds/2019.1-unbrake #40: Criar formulário para configurações do ensaio

joaorobson, 2019-03-31:
## Descrição

Esta issue é referente ao desenvolvimento do restante do formulário para configurações gerais do simulador. A issue para criar parte inicial do form é a #9.

## Tarefas

- [ ] Utilizar [redux-form-material-ui](https://github.com/erikras/redux-form-material-ui/tree/5.0) para criar os campos editáveis.
- [ ] Alimentar os campos editáveis com as informações contidas no arquivo que pode ser inserido usando os campos criados na issue #9.
- [ ] Utilizar Material-UI para criar o restante dos componentes.

## Informação adicional

Após ler o arquivo que possui a extensão ".ini", as informações devem ser passadas aos seus respectivos campos. Para realizar isso, deve-se utilizar uma biblioteca que converta o que há no arquivo para um objeto JS. A biblioteca sugerida é a [ini](https://www.npmjs.com/package/ini).

Abaixo, página correspondente no software utilizado atualmente:
[image]

GabrielTiveron, 2019-04-02:
A tela de configuração com sua respectiva rota na issue #7 não seria uma dependência também?

tmcstiago, 2019-04-03:
Sim @GabrielTiveron. Por isso é a ideia que eu entregue logo esta Issue.

closed 2019-04-08 by tmcstiago