fga-eps-mds/2019.1-unbrake #47: Realizar integração da autenticação no front-end

joaorobson, 2019-04-01:
## Descrição

Realizar autenticação utilizando página de login criada na issue #8.
## Tarefas

- [ ] Realizar a integração da autenticação com a página de login.
- [ ] Criar um mecanismo que seja seguro para prover o token para o frontend.

## Informação adicional

A maneira mais fácil de prover o token para o React é utilizando localStorage ou sessionStorage (algo como uma "memória local" acessível pelo JS). Porém, essas formas não são recomendáveis pela possibilidade de serem suscetíveis à ataques, como explicado aqui. Nesse mesmo link é explicado como realizar essa tarefa de maneira segura utilizando cookies.

joaorobson, 2019-04-14:
Os cookies foram utilizados para armazenar o token, porém a solução ainda deve ser aprimorad/testada posteriormente, pois o cookie deve poder ser acessado apenas pelo servidor da aplicação e por enquanto esse acesso não está sendo realizado ainda.

closed 2019-04-14 by joaorobson

Bumbleblo, 2019-04-16:
Colocar estimativa nessa issue