fga-eps-mds/2019.1-unbrake #48: Decisão sobre protocolo de comunicação do gráfico

Bumbleblo, 2019-04-01:
## Descrição

Abrir um discussão sobre qual protocolo de comunicação utilizar para comunicar o Front-End com o servidor da placa em tempo real.

## Tarefas

- [ ] Discutir sobre mqtt
- [ ] Testar o MQTT vs o UDP
- [ ] Procurar protocolos alternativos

## Informação adicional

É importante que o protocolo de comunicação entregue com pouca latência o gráfico mesmo em rede congestionadas, portanto temos que procurar a melhor solução para entrega de pacotes. Veja que não é preciso que os pacotes cheguem na ordem.

icaropires, 2019-04-02:
Aqui está um [link](https://www.pubnub.com/blog/2016-05-10-getting-started-with-iot-in-realtime/) interessante da PubNub sobre possíveis soluções.

icaropires, 2019-04-02:
Outro [material](https://www.pubnub.com/blog/project-eon-open-source-javascript-framework-for-realtime-dashboard-charts-and-maps/) interessante especificamente sobre dashboards, também da PubNub.

icaropires, 2019-04-02:
Aqui está uma [solução](https://itnext.io/javascript-real-time-visualization-of-high-frequency-streams-d6533c774794) semelhante ao que queremos utilizando Vue e MQTT.

icaropires, 2019-04-02:
Aqui estão algumas [opções](https://www.infoworld.com/article/2972143/real-time-protocols-for-iot-apps.html) possíveis de protocolo.

icaropires, 2019-04-02:
Uma opção a ser investigada é o ecossistema [Gun](https://github.com/amark/gun) de ferramentas.

Bumbleblo, 2019-04-02:
Segundo a documentação do [mosquito](https://mqtt.org/tag/mosquitto) o mqtt necessariamente precisa de um broker, portanto podemos utilizar qualquer um com suporte. Sugiro o [Rabbitmq](https://www.rabbitmq.com/) mais informações nos links:

- [mqtt web rabbitmq plugin](https://www.rabbitmq.com/web-mqtt.html)
- [mqtt plugin ](https://www.rabbitmq.com/mqtt.html)

Bumbleblo, 2019-04-03:
Novos requisitos mostraram que os pacotes teem a necessidade de chegarem na ordem para o gráfico no React uma vez que a variável tempo é utilizada para desenha-lo.

Bumbleblo, 2019-04-03:
@icaropires, @ViniciusBernardo e @joaorobson gostariam da opnião de vocês sobre essas escolhas.

icaropires, 2019-04-05:
Como eu pude observar nos links associados, a comunidade utiliza o MQTT como uma solução robusta para o problema. Então acredito que ele como protocolo vá atender bem.

> Segundo a documentação do [mosquito](https://mqtt.org/tag/mosquitto) o mqtt necessariamente precisa de um broker, portanto podemos utilizar qualquer um com suporte. Sugiro o [...]

Não acho que seja assunto dessa issue escolher a ferramenta que será utilizada, apenas o protocolo. Na issue de integração da ferramenta poderemos discutir sobre a ferramenta se for desejo do time.

closed 2019-04-06 by Bumbleblo