fga-eps-mds/2019.1-unbrake #58: Analisar viabilidade do uso do GraphQL

Bumbleblo, 2019-04-03:
## Descrição

Ler a documentação e analisar as features e suporte do [GraphQL](https://graphql.github.io/)

## Tarefas

- [ ] Analisar as features que permite as substituição do GET
- [ ] Se existe uma forma de trabalhar com POST
- [ ] Se todas as linguagens do projetos tem suporte pelo GraphQL
- [ ] Analisar esquema de autenticação em cima das query do GraphQL

## Informação adicional

[RTFM](https://en.wikipedia.org/wiki/RTFM) in [docs](https://graphql.github.io/)

Bumbleblo, 2019-04-03:
Sobre a substituição do GET temos essa documentação que mostra que nesse quesito o GraphQL tem muitas vantagens acima do REST.

link: https://graphql.github.io/learn/queries/

Bumbleblo, 2019-04-03:
Para inserir dados no banco o GraphQL fornece as [mutations](https://graphql.github.io/learn/queries/#mutations)

Bumbleblo, 2019-04-03:
O GraphQL tem suporte para o Python e para o Go porém não trabalha com o React diretamente.

fonte: https://graphql.github.io/code/

Bumbleblo, 2019-04-03:
> O GraphQL tem suporte para o Python e para o Go porém não trabalha com o React diretamente.
> 
> fonte: https://graphql.github.io/code/

Isso não é um problema uma vez que o React se comunica por requisições sobre a API

Bumbleblo, 2019-04-03:
A forma que a documentação deixa explicita para autenticação é utilizando uma condicional em cima do tipo do objetivo GraphQl, um problema é a replicação de código uma vez que em cada tipo nós temos que criar uma função de autenticação. Porém a própria documentação tem uma forma de minimizar isso.

fonte: https://graphql.org/learn/authorization/

Bumbleblo, 2019-04-03:
> A forma que a documentação deixa explicita para autenticação é utilizando uma condicional em cima do tipo do objetivo GraphQl, um problema é a replicação de código uma vez que em cada tipo nós temos [...]

eis um exemplo

``` javascript
//Authorization logic lives inside postRepository
var postRepository = require('postRepository');

var postType = new GraphQLObjectType({
  name: ‘Post’,
  fields: {
    body: {
      type: GraphQLString,
      resolve: (post, args, context, { rootValue }) => {
        return postRepository.getBody(context.user, post);
      }
    }
  }
});
```

Bumbleblo, 2019-04-03:
Espero avaliação do @joaorobson, @ViniciusBernardo e @icaropires para estabelecer os proximos passos. Fazemos algum benchmark ? Isso resolve o nosso problema no escopo da API ? Vamos discutir esses aspectos pessoal!

icaropires, 2019-04-03:
Como com certeza ele resolve muitos problems de GET, que nós vamos precisar. Podemos integrá-lo e depois se ele não resolver todos problemas, podemos complementar com Rest

Bumbleblo, 2019-04-03:
> Como com certeza ele resolve muitos problems de GET, que nós vamos precisar. Podemos integrá-lo e depois se ele não resolver todos problemas, podemos complementar com Rest

acredito que seja uma forma de integrar, vamos fazer isso até a R1 ou depois dela ?

icaropires, 2019-04-05:
Seria antes da R1, se não: ou não entregaríamos a nada da API em funcionamento ou faríamos em Rest e depois apagaríamos, o que não faz muito sentido.

joaorobson, 2019-04-05:
Acredito que seria factível até a R1 usarmos o GraphQL. Ele facilita muito a questão de realizar queries via API e pelos exemplos dados pelo @Bumbleblo a questão da autenticação é resolvida de maneira simples também.

closed 2019-04-06 by Bumbleblo