fga-eps-mds/2019.1-unbrake #62: Melhorar o tema da página do projeto

Bumbleblo, 2019-04-05:
## Descrição

Colocar um tema pre-montado na página do projeto

## Tarefas

- [ ] Escolher um tema visualmente aceitável
- [ ] Realizar as configurações necessárias no pelican para que configure o tema corretamente

## Informação adicional

https://docs.getpelican.com/en/stable/pelican-themes.html

Bumbleblo, 2019-04-12:
Os melhores links que encontrei com temas para o Pelican estão nesse [site](http://pelicanthemes.com/) porém o único que atende as nossas necessidades não está compatível com a versão que estamos utilizando.

icaropires, 2019-04-12:
Atualizar é uma opção?

Bumbleblo, 2019-04-14:
Teríamos que entender a fundo como elaborar o tema e provavelmente vamos ter que trabalhar diretamente no html css.

Bumbleblo, 2019-04-14:
Vou utilizar esse [tema](https://github.com/Parbhat/pelican-blue), ele vai conseguir resolver o problema com os links que estamos tendo agora.

Bumbleblo, 2019-04-14:
Essa é a versão original com esse tema, a partir dele tem como alterar o template para adicionar algumas outras coisas porém para um versão inicial acredito que está bom.

[image]

Bumbleblo, 2019-04-14:
Já foi realizado o deploy

closed 2019-04-14 by Bumbleblo