fga-eps-mds/2019.1-unbrake #64: Configurar ferramentas de folha de estilo e linters para o backend

icaropires, 2019-04-06:
## Descrição

Configurar ferramentas de folha de estilo e linters para o backend para uso local pelos desenvolvedores

## Tarefas

- [ ] Escolher ferramenta de folha de estilo
- [ ] Integrar ferramenta de folha de estilo
- [ ] Escolher ferramenta linter
- [ ] Integrar ferramenta linter
- [ ] Integrar configurações entre linter e folha de estilo para que não discordem
- [ ] Ambas as ferramentas devem ser integradas no esquema de git hooks
- [ ] O chamado dos git hooks devem ser executados em ambiente dockerizado e sem ser necessário intervenção por parte do desenvolvedor
- [ ] Atualizar integração contínua para que execute as novas ferramentas integradas

## Informação adicional

icaropires, 2019-04-06:
A ferramenta utilizada como checador da folha de estilo será a `flake8`, que também é um linter. Ela é a escolhida por ser amplamente aceita pela comunidade há bastante tempo.

icaropires, 2019-04-06:
Na minha opinião o flake8 é uma excelente ferramenta para casos de uso mais simples. Mas opto por também utilizar a ferramenta [pylint](https://www.pylint.org/) para checar vários aspectos de linting que a flake8 não detecta: como falta de documentação, falta de atribuição, regex para nomes de variáveis, detecção de erros, entre outros (sim).

Como é um projeto é em django, há também uma extensão que facilita a personalização da configuração do pylint para o django, é a [pylint-django](https://github.com/PyCQA/pylint-django), também integrarei ela.

icaropires, 2019-04-06:
Eu poderia mas não vou, exceto talvez sob motim, ignorar os avisos do pylint sobre falta de documentação em código. Pois, é uma boa prática e ajudará os outros desenvolvedores e até o próprio implementador mais tarde.

Aos desenvolvedores, @medeiroslucas, @tmcstiago, @leticiadesouza, @VictorLeviPeixoto e @GabrielTiveron, por favor consultar a [pep 257](https://www.python.org/dev/peps/pep-0257/).

joaorobson, 2019-04-06:
[image]

PR #66 (2019-04-07): Configurar ferramentas de checkagem do backend
Closes #64 

Cumpre o estabelecido na issue e o discutido nos comentários da issue. Checa folha de estilo e executa análise estática antes do commit e executa os testes antes do push, além de estar tudo integrado no CI.

[image]
closed 2019-04-07 by joaorobson