fga-eps-mds/2019.1-unbrake #71: Criar Roadmap de Devops

icaropires, 2019-04-07:
## Descrição

Criar Roadmap de devops, detalhando o que pretende-se alcançar e quando.

## Tarefas

- [ ] Mapear o que já feito e as intenções
- [ ] Planejar o que fazer até a R1 e os motivos
- [ ] Colocar no github pages

## Informação adicional

RochaCarla, 2019-04-08:
quais sao os criterios de aceitação?!!!

icaropires, 2019-04-08:
@RochaCarla, essa issue ainda será evoluída quando eu souber melhor o que vou fazer.

Mas estamos mapeando os critérios de aceitação também como tasks. O revisor do PR avaliará se todas as tasks realmente foram cumpridas. Tem problema fazer dessa forma?

icaropires, 2019-04-20:
Adicionei o template do diagrama de roadmap e a parte das tarefas de devops até a R1. Ele pode ser acessado aqui: [diagrama](https://fga-eps-mds.github.io/2019.1-unbrake/roadmaps/roadmap-papeis/).

icaropires, 2019-04-20:
Se tiver tudo ok, pode fechar @joaorobson

closed 2019-04-24 by joaorobson