fga-eps-mds/2019.1-unbrake #78: Entender o funcionamento da linguagem Go

Bumbleblo, 2019-04-12:
## Descrição

Entender a linguagem Go para o microserviço servidor

## Tarefas

- [ ] Realizar o tour do Go
- [ ] Documentar links úteis nesta issue 
- [ ] Escrever uma função que realiza uma requisição GET 
- [ ] Realizar testes nessa função

## Informação adicional

[Tem a documentação oficial](https://golang.org/) e também pode usar a sua coragem

Bumbleblo, 2019-04-12:
O tour do Go pode ser encontrado nesse [link](https://tour.golang.org/welcome/2)

Bumbleblo, 2019-04-12:
@icaropires existem algumas ferramentas de análise estática de código nativas no Go, nesse [link](https://github.com/golang/go/wiki/CodeTools) da wiki tem mais informações detalhadas. Se puder dar uma olhada e realizar as integrações destas ferramentas no ambiente de desenvolvimento.

icaropires, 2019-04-12:
> @icaropires existem algumas ferramentas de análise estática de código nativas no Go, nesse [link](https://github.com/golang/go/wiki/CodeTools) da wiki tem mais informações detalhadas. Se puder dar uma olhada e realizar as integrações destas ferramentas no ambiente de desenvolvimento.

@Bumbleblo , sim, esse foi um dos motivos pelos quais achei que seria legal trabalhar com Go, ela é muito bem resolvida nesse sentido de estilo e linting.

Vou adicionar essa integração como tarefa para a próxima sprint. Só que tenho que estudar melhor a questão do ambiente no windows antes.

Bumbleblo, 2019-04-13:
Ótimo, estou deixando os links úteis na issue para ficar mais fácil encontrar esse conteúdo depois

icaropires, 2019-04-13:
Ow integração, vc funciona direito agora?

icaropires, 2019-04-13:
e se eu mencionar alguém @Bumbleblo ?

Bumbleblo, 2019-04-14:
Para criar o servidor http vou tentar utilizar o [pacote padrão do Go](https://golang.org/pkg/net/http/)... não sei se é a melhor maneira mas vou tentar deixar o sistema bem simples

Bumbleblo, 2019-04-14:
A melhor alternativa é fazer o que o Go chama de custom server, isso evita que usemos o servidor padrão do go para tudo.

Bumbleblo, 2019-04-14:
Para a questão do http.Handle eu não achei uma explicação satisfatória na documentação oficial, porém, nesse blog tem um código mostrando como usar o Handle, talvez seja melhor dar uma olhada na FuncHandle também porém somente o handle padrão já resolve o nosso problema.

Bumbleblo, 2019-04-15:
Consegui escrever um código base que sobe um servidor http.

Bumbleblo, 2019-04-15:
```Go
package main

import "net/http"
import "fmt"

func callback(w http.ResponseWriter, r *http.Request){
    fmt.Fprintf(w, "Hello World")
}

func main(){

    http.HandleFunc("/porra", callback)
    http.ListenAndServe(":8080", nil)
}
```

Bumbleblo, 2019-04-15:
Tentei fazer com o servidor customizado mas a resposta prática é: tem zilhões de maneira de fazer a mesma coisa e o modo mais fácil e este acima. Caso queira fazer vários endpoints precisa fazer um Mux e depois adicionar um endpoint para cada coisa que você precisar.

Bumbleblo, 2019-04-15:
Para realizar o teste tem essa [lib](https://golang.org/pkg/testing/). Para realizar o mock no http existe essa o pacote [httptest](https://golang.org/pkg/net/http/httptest/#NewServer).

Bumbleblo, 2019-04-16:
Aprendam a trabalhar com interfaces, praticamente tudo em Go é interface e entender isso é crucial pra entender alguma coisa da documentação. Tem informação sobre isso em um módulo do tour que está acima.

Bumbleblo, 2019-04-16:
Estou bem travado na parte do teste do servidor. Por vários motivos:

* Não consegui separar o arquivo de teste e de o código (sei a solução mas é trabalhosa e esta detalhada nesse [link](https://golang.org/doc/code.html#ImportPaths))
* O Go ainda não está executando meus testes realizados mesmo que esteja no mesmo arquivo

Bumbleblo, 2019-04-16:
Vou explorar essa [documentação](https://golang.org/doc/code.html#Testing) que exemplifica como realizar o teste. Isso não abrange o httptest porém já é alguma coisa

Bumbleblo, 2019-04-16:
Consegui resolver todos os meus problemas com essa documentação acima. Ela abrangia exatamente a parte que fica mais obscura na definição do pacote.

Bumbleblo, 2019-04-16:
O teste ainda não está muito bom mas como é somente para aprendizado acredito que já atinge o objetivo, para testar melhor deveria ver se o corpo da requisição tem a mensagem que estou querendo. O código final ficou assim:

http server coloquei o nome do arquivo de **program.go**:

```Go
package know

import "fmt"
import "net/http"

func Callback(w http.ResponseWriter, r *http.Request){
    fmt.Fprintf(w, "Hello World")
}

func main(){

    http.HandleFunc("/test", Callback)
    http.ListenAndServe(":8080", nil)
}

```
O teste desse arquivo, perceba que tem que ter o sufixo _test em cada arquivo de testes:

```Go
package know

import "testing"
import "fmt"
import "net/http/httptest"

func TestCallback(t *testing.T){
    req := httptest.NewRequest("GET", "localhost:8080", nil)
    w := httptest.NewRecorder()
    Callback(w, req)

    resp := w.Result().StatusCode
    fmt.Println(resp)

```

Depois de executar o comando **go test** esse foi a saída do console

[image]

Bumbleblo, 2019-04-16:
@icaropires provavelmente essas funções da documentação serão úteis para realizar a integração continua:

https://golang.org/pkg/testing/#Coverage
https://golang.org/pkg/testing/#CoverMode

Bumbleblo, 2019-04-16:
@icaropires provavelmente essas funções da documentação serão úteis para realizar a integração continua:

https://golang.org/pkg/testing/#Coverage

closed 2019-04-16 by Bumbleblo