fga-eps-mds/2019.1-unbrake #84: Avaliar melhor solução para integração inicial do Go com o React (MQTT vs WebSocket)

joaorobson, 2019-04-14:
## Descrição

Realizar um levantamento sobre qual solução é mais viável para realizar uma integração inicial entre os serviços providos pelo Go e o front-end em React.

## Tarefas

- [ ] Pesquisar o uso das duas alternativas para exibir dados em tempo real no React.
- [ ] Avaliar se WebSocket oferece recursos de ordenação dos dados.
- [ ] Caso a avaliação do ponto anterior tiver resultado positivo, avaliar qual das duas opções (MQTT vs WebSocket) oferece uma simplicidade/rapizez maior na integração.

Bumbleblo, 2019-04-16:
foi sugerido pelo time de deixarmos essa issue de lado e implementarmos apenas com um socket para a primeira release. Deixamos a cargo do @joaorobson decidir qual a melhor maneira de realizar essa integração para a primeira release.

joaorobson, 2019-04-22:
Como dito pelo @Bumbleblo , por conta do prazo curto, a solução utilizada para a primeira release se baseará em WebSocket. Como explicado [aqui](https://developer.mozilla.org/pt-BR/docs/WebSockets/Writing_WebSocket_servers), WebSocket consiste em uma aplicação TCP. Esse protocolo, como especificado na [RFC 793](https://tools.ietf.org/html/rfc793), garante ordenação dos dados transmitidos. Claramente, assegurar isso custa um preço, nesse caso, latência. Como se trata apenas de um protótipo inicial para validar o projeto e exemplos bastante similares ao nosso caso existem, como [esse aqui](https://medium.com/@oxygenna/real-time-cryptocurrency-visualisations-using-react-websockets-and-chartjs-e4a76132a862), inicialmente optaremos pelo WebSocket

closed 2019-04-22 by joaorobson