fga-eps-mds/2019.1-unbrake #91: Criar Documento de visão

joaorobson, 2019-04-15:
## Descrição

Criar documento de visão do projeto.

## Tarefas

- [ ] Criar documento de visão no Google Drive.
- [ ] Colocar documento no GitHub Pages.

## Informação adicional

Podem se basear no [documento](https://github.com/fga-eps-mds/2017.1-PlataformaJogosUnB/wiki/Documento-de-Vis%C3%A3o) do UnBGames.

Bumbleblo, 2019-04-25:
Gostaria que MDS conversasse sobre a visão de cada um sobre o projeto nessa issue. O documento de visão tem praticamente a função de unificar a visão da equipe que está trabalhando no projeto então é extremamente necessária essa discussão.

Bumbleblo, 2019-04-25:
@joaorobson @icaropires @Bumbleblo @ViniciusBernardo coloquem a visão de vocês também para incentivar a discussão!

medeiroslucas, 2019-04-25:
Minha percepção do projeto é que estamos desenvolvendo uma plataforma que erá controlar, receber os dados e gerar análises de uma bancada de teste de frenagem automotivo. Minha principal dúvida é em relação a como devemos tratar o projeto. Se estamos lidando com uma plataforma feita por encomenda que atende requisitos específicos de um único cliente, visto que a utilização correta do software depende de elementos únicos da bancada propriamente dita. Ou se a plataforma que está sendo desenvolvida poderá ser implantada em outras bancadas realizando-se as devidas alterações

medeiroslucas, 2019-04-25:
Outro ponto pouco claro para mim são os privilégios que o administrador tem em relação ao usuário regular

tmcstiago, 2019-04-26:
Por se tratar de um software muito específico, não temos concorrentes conhecidos no mercado, uma vez que quando as empresas tem necessidade de ter o mesmo, compram o software por encomenda.

Na minha visão, acredito que não temos o Evandro como único cliente, e sim como um potencial cliente entre os vários que podem utilizar o software. Entendo isso, pois mesmo fazendo o software especifico para rodar no projeto do Galpão e interagindo com um hardware especifico, temos o direito de comercialização do software, podendo fazer adaptações para hardware especifico dos novos clientes.

tmcstiago, 2019-04-27:
Para facilitar a edição do documento por todos os participantes, estamos usando a plataforma do HackMD. O mesmo se trata de uma aplicação parecida com o Google Docs, no qual todos os usuários podem realizar a edição online do arquivo já utilizando a linguagem de marcação Markdown.

O documento pode ser acessado através do link https://hackmd.io/9EQYdYFAQqqhDUodrf6MIg

Bumbleblo, 2019-04-27:
> Outro ponto pouco claro para mim são os privilégios que o administrador tem em relação ao usuário regular

Nesse ponto aqui @medeiroslucas a resposta é: O usuário administrador é aquele que tem privilégio de alterar os arquivos de ensaio base.

Bumbleblo, 2019-04-27:
Concordo com o @tmcstiago e @medeiroslucas no tocante ao projeto não ser inteiramente direcionado a único circuito embarcado. Acredito que deveríamos direcionar a realização de testes genéricos de frenagem, no entanto, isso irá nos direcionar a um estudo profundo das metodologias imposta nas ISO.

Bumbleblo, 2019-04-27:
Na minha visão é um software de apoio a realização dos teste de frenagem que se propõe a realizar tanto os ensaios quanto a realizar as principais análises que um técnico precisa para tomar suas decisões.

leticiaarj, 2019-04-27:
Na minha opinião estamos desenvolvendo um projeto relativamente grande, em relação aos outros e com um nível de dificuldade um pouco maior. Por isso, acho que demoramos muito pra entender o projeto e para compreender as tecnologias ultilizadas. Todavia, recebemos muito apoio dos alunos de EPS, que estão nos coordenando. Por isso, acho que o projeto vai ser de grande auxílio para o professor Evandro e complementarmente para toda a comunidade acadêmica.

joaorobson, 2019-04-27:
Acredito que o prof. Evandro é um cliente e que o foco do projeto é resolver o problema dele. A posteriori, caso exista algum modelo parecido de ensaios veiculares e que utilize um controlador eletrônico similar que transmite os dados coletados via comunicação serial, podemos pensar em uma solução mais genérica. Por enquanto, ao meu ver, o problema real e atual a ser resolvido é o proposto pelo prof. Evandro. Não vejo problemas em resolver um problema muito específico de uma forma que é bastante específica também, contanto que o objetivo de matar o problema seja atingido.

GabrielTiveron, 2019-04-28:
A linguagem utilizada para a leitura dos dados do simulador será Go?

closed 2019-04-29 by tmcstiago

joaorobson, 2019-05-01:
> A linguagem utilizada para a leitura dos dados do simulador será Go?

Sim