fga-eps-mds/2019.1-unbrake #107: Criar Roadmap dos papéis

icaropires, 2019-04-20:
## Descrição

Deve ser criado um diagrama de roadmap ilustrando o que cada gerente pretende realizar de tarefa do seu papel até a R1.

## Tarefas

- [ ] Criar template
- [ ] Adicionar tarefas de DevOps
- [ ] Adicionar tarefas de ScrumMaster
- [ ] Adicionar tarefas de Arquitetura
- [ ] Adicionar tarefas de Engenheiro de produto
- [ ] Ao final o diagrama deve estar disponível no github pages do projeto

## Informação adicional

icaropires, 2019-04-20:
O template foi criado e pode ser visualizado [aqui](https://docs.google.com/spreadsheets/d/1Wk9N9ItT7frXfL9IvOn7ZVAOPFcqtAwj6WKRGJUEUuU/edit?usp=sharing) mas só pode ser alterado por membros da aquipe.

Bumbleblo, 2019-04-21:
Esse épico está replicado com o épico #88

icaropires, 2019-04-21:
Realmente, podemos manter somente esse que está mais completo?

joaorobson, 2019-05-01:
Posso fechar esse épico? O @Bumbleblo não adicionou no template, mas em uma aba separada. Tudo ok por você, @icaropires ?

icaropires, 2019-05-01:
Acho que você @joaorobson e o @ViniciusBernardo podem decidir isso.

Bumbleblo, 2019-05-08:
Como os roadmaps iniciais foram criados estou fechando esse épico. Atualizações e melhorias podem ser adicionados posteriormente referente a épicos da release 2.

closed 2019-05-08 by Bumbleblo