fga-eps-mds/2019.1-unbrake #120: Telas de login e cadastro não estão acessíveis a um usuário

icaropires, 2019-04-27:
## Descrição do problema

Ao entrar no sistema não 

## Passos para reprodução

Passos para reproduzir o comportamento.

1. Vá para...
2. Click em...
3. ...

### Comportamento atual

O que está acontecendo atualmente

### Comportamento esperado

Qual é o comportamento esperado pelo sistema

## Contextos, prints e informações adicionais

Se for aplicável, adicione gifs, imagens ou qualquer outro detalhe de
enriqueça a descrição do problema.

## Possível Solução

Caso você tenha algum sugestão de onde começar a procurar o problema ou uma
solução para ele escreva aqui.

icaropires, 2019-04-27:
Aparentemente também está ocorrendo outro erro relacionado com as rotas, porque com a build de produção não é possível nem acessar os paths manualmente.

Checar referência:
https://stackoverflow.com/questions/27928372/react-router-urls-dont-work-when-refreshing-or-writing-manually

tmcstiago, 2019-04-27:
Para resolver este problema, a página de Login foi definida como página inicial do sistema e foi a funcionalidade de incluir novo usuário foi incluída em uma das opções do menu lateral, além de ter o redirecionamento através do botão "Cadastrar" da página de Login.

PR #122 (2019-04-28): Ajustando rotas para as páginas Login e Sign Up
## Problema/Funcionalidade

O sistema não tinha nenhum botão para acesso a página de Login e para criar um novo usuário.

### Como foi implementado

Para que posse possível acessar estas duas páginas, foi alterado as rotas dos arquivos "Routes.jsx" e "SideBarMenu.jsx".
Após estas devidas alterações, é possível acessar a página de Login através da página inicial e é possível acessar a pagina para criar novo usuário através do último componente do menu ("Sign Up") e pelo botão cadastrar da página de Login.

### Autores

@tmcstiago 

## Issue

resol [... cut]
closed 2019-04-28 by icaropires