fga-eps-mds/2019.1-unbrake #127: Tela de login não some

icaropires, 2019-04-28:
## Descrição do problema

Ao se visitar a tela de login e depois ir para outra, a de login continua em exibição, mas abaixo na tela.

## Passos para reprodução

1. Subir serviço do frontend
2. Visitar localhost:3000/login
3. Clicar em outra página a partir do menu
4. Rolar a tela para baixo

### Comportamento atual

A tela de login não desaparece ao trocar de janela.

### Comportamento esperado

Que a janela de login desaparecesse depois de sair da janela de login.

## Contextos, prints e informações adicionais

[image]

## Possível Solução

tmcstiago, 2019-04-28:
Este bug também acontece quando é acessada a url "/signup", tanto pela url quanto pelo botão "Cadastrar" da página de Login.

Também é importante dizer se for acessado a página de Login através do localhost:3000, o bug mencionado não acontece.

icaropires, 2019-04-30:
Esse bug foi resolvido no PR #141

closed 2019-04-30 by icaropires