fga-eps-mds/2019.1-unbrake #147: Conteúdo do token não está sendo checado

icaropires, 2019-04-30:
## Descrição do problema

O conteúdo do token de sessão não está sendo checado, então qualquer um pode ser considerado logado na aplicação.

## Passos para reprodução

1. Entre no modo de desenvolvedor do navegador
2. Crie um cookie manualmente com a chave 'Token' e qualquer valor

### Comportamento atual

Usuário considerado logado

### Comportamento esperado

O token deveria ser considerado inválido

## Contextos, prints e informações adicionais

## Possível Solução

GabrielTiveron, 2019-05-05:
Existe alguma duração padrão para os cookies? Eu coloquei para expirar em 1 hora

icaropires, 2019-05-05:
Acho que uma abordagem legal seria renovar ele por n (10? 5?) minutos a cada requisição. Só para não deslogar um cara que ta mexendo e sim quem ficar ausente.

PR #153 (2019-05-08): Checar conteudo token
# Fazer logica de verificação e validação do token 
### Problema
O token não era verificado, sendo assim tendo qualquer valor era aceito pelo sistema
### Como foi implementado 
É feito um request para validar o token existente no navegador, o mesmo tem uma validade de um dia. 
### Autores 
@GabrielTiveron 
### Issue
resolves #147
closed 2019-05-08 by joaorobson