fga-eps-mds/2019.1-unbrake #149: Consertar issues apontadas pelo Code Climate no back-end

joaorobson, 2019-05-05:
## Descrição

Consertar issues apontadas [aqui](https://codeclimate.com/github/fga-eps-mds/2019.1-unbrake/issues?status%5B%5D=&status%5B%5D=open&status%5B%5D=confirmed&language_name%5B%5D=Python)

## Tarefas

- [ ] Avaliar quais issues são possíveis arrumar e apontar/comentar aqui na issue quais não são;
- [ ] Arrumar issues.

## Informação adicional

medeiroslucas, 2019-05-07:
Pesquisando sobre a viabilidade de remover as duplicações de código no graphQL encontrei [esse post](https://medium.freecodecamp.org/five-common-problems-in-graphql-apps-and-how-to-fix-them-ac74d37a293c) que dá uma solução utilizando o Postgre. Como não entendi completamente essa solução quero propor uma discussão sobre a viabilidade de utiliza-la

joaorobson, 2019-05-07:
Pode deixar essas issues sem arrumar porque a própria estrutura do GraphQL exige que os métodos sejam assim. Sendo assim, iremos ignorar esses arquivos na checagem do Code Climate por enquanto.

PR #152 (2019-05-08): Consertar issues apontadas pelo Code Climate no back-end
# Resolver problemas do Code Climate 
### Problema
Trechos com alta complexidade ciclomática nos testes de backend
### Como foi implementado 
Substituindo os repetidos assets para cada campo do json por um único assert que checa tudo
### Autores 
@medeiroslucas 
@leticiaarj 
### Issue
resolves #149
closed 2019-05-08 by joaorobson