fga-eps-mds/2019.1-unbrake #163: Permitir somente administrador cadastrar usuários

joaorobson, 2019-05-12:
## Descrição

Bloquear tela de cadastro para usuários comuns e habilitá-la para o admin.

## Tarefas

- [ ] Enviar atributo is_superuser para o frontend e salvar informação no localStorage;
- [ ] Utilizar AuthorizedRoute com a prop permission para validar se usuário é admin ou não.

## Informação adicional

O atributo is_superuser pode ser enviado juntamente com o token, o que evitaria realizar outra requisição para acessar essa informação.

GabrielTiveron, 2019-05-16:
Haverá apenas um usuário admin?

tmcstiago, 2019-05-16:
Acredito que para uma melhor utilização do Software, o ideal seria ter vário usuários com permissão de admin

VictorLeviPeixoto, 2019-05-17:
Acredito que não, tendo em vista o poder do mesmo.
Na prática nenhum aluno deveria ter tal acesso, somente alguns professores isso se passar do próprio Evandro

PR #173 (2019-05-19): Admin com cadastro de usuario
# Permite apenas ao admin cadastrar usuários

## Problema/Funcionalidade

É permitido apenas ao usuário administrador o cadastro de outros usuários.

### Como foi implementado

Foi utilizada o atributo "isSuperuser" para filtrar o tipo de usuário em questão para que se saiba quem tem a permissão de cadastrar outro usuário

### Possíveis melhorias

### Autores

@GabrielTiveron  @VictorLeviPeixoto 

### Issue

resolves #163
closed 2019-05-19 by joaorobson