fga-eps-mds/2019.1-unbrake #176: Criar formulário de calibração de força

joaorobson, 2019-05-19:
## Descrição

Criar formulário para calibração da força

## Tarefas

- [ ] Criar formulário utilizando redux-form-material-ui se baseando na estrutura utilizada no formulário de configuração;
- [ ] Criar botão para submissão das informações para o back-end;
- [ ] Integrar gráfico na página.

## Informação adicional

tmcstiago, 2019-05-22:
Para auxiliar o desenvolvimento do formulário, segue abaixo imagem deste formulário no sistema atual.

[image]

tmcstiago, 2019-05-23:
Conforme informado na #178 , não será incluído o botão de upload nos formulários secundários da calibração.

closed 2019-05-26 by tmcstiago