fga-eps-mds/2019.1-unbrake #178: Criar formulário de calibração de vibração

joaorobson, 2019-05-19:
## Descrição

Criar formulário para calibração da vibração

## Tarefas

- [ ] Criar formulário utilizando redux-form-material-ui se baseando na estrutura utilizada no formulário de configuração;
- [ ] Criar botão para submissão das informações para o back-end;
- [ ] Integrar gráfico na página.

## Informação adicional

tmcstiago, 2019-05-21:
Para facilitar o trabalho durante o desenvolvimento da Issue, segue imagem desta página no sistema anterior:

[image]

tmcstiago, 2019-05-21:
O campo Canal de aquisição deve mesmo ter o campo bloqueado?
Entendo que os campos que encontram-se bloqueados são os que recebem informações do sistema, porém o campo "Canal de aquisição" é inserido pelo o usuário através do upload de arquivo.

joaorobson, 2019-05-21:
Sim, deve ser um campo read-only

tmcstiago, 2019-05-21:
Em discussão sobre a melhor maneira de utilização do software, foi decidido que não teremos botão para submeter as informações nesta página.
O botão para submeter as informações ficará apenas na aba geral da parte da calibração

VictorLeviPeixoto, 2019-05-21:
Essa decisão se aplica para todas as issues relacionadas a criação de formulários para a calibração?

tmcstiago, 2019-05-22:
> Essa decisão se aplica para todas as issues relacionadas a criação de formulários para a calibração?

Sim. O botão para submit será incluído na aba Geral da calibração

closed 2019-05-22 by joaorobson