fga-eps-mds/2019.1-unbrake #194: O usuário não é redirecionado pra página de Login caso não esteja logado

tmcstiago, 2019-05-26:
## Descrição do problema

Em determinadas situações o usuário não é redirecionado para a página de login caso esteja deslogado do sistema.

## Passos para reprodução

Passos para reproduzir o comportamento.

1. http://unbrake-hom.ml/;
2. Faça login na aplicação;
3. Espere o token expirar ou faça a exclusão dos cookies da página;
4. Recarregue a página novamente.

### Comportamento atual

Caso o usuário esteja em uma página diferente da página "/login", a página não mostra nenhuma informação na tela.

### Comportamento esperado

Independente da página que o usuário se encontre, o mesmo deve ser redirecionado para a "/login" caso o mesmo não esteja logado.

## Possível Solução

O redirecionamento não funciona apenas se tiver informações localizadas no local storage.
Para a solução, acredito que no momento de verificação do login não seja verificado as informações presentes no local storage.

tmcstiago, 2019-05-29:
Apesar do PR #202 que informa que foi resolvido esta Issue, o problema ainda persiste.

Além dos erros expostos nesta Issue, após excluir o cookie não é possível acessar nem a página de "/login" caso ainda exista a informação no local Storage.

VictorLeviPeixoto, 2019-05-29:
Não tinha relacionado o problema corretamente, devido isso, irei fazer essa issue junto com a minha atual no mesmo PR, tendo em vista o pequeno tamanho do escopo da mesma.

tmcstiago, 2019-06-11:
Esta issue está sendo reaberta devido ao problema não ter sido resolvido na branch informada.

PR #202 (2019-05-29): Consertar problemas na autentificação
## Problema/Funcionalidade

Foi criado um formulário para a parte de velocidade dentro do módulo de calibração

### Como foi implementado

- Foram modificados os arquivos Auth.jsx e Routes.jsx.

### Autores

@VictorLeviPeixoto 

### Issue

Resolves #196 
Resolves #195 
Resolves #194 
Resolves #162
closed 2019-06-19 by tmcstiago