fga-eps-mds/2019.1-unbrake #241: Página configuration não é carregada

icaropires, 2019-06-11:
## Descrição do problema

Página configuration não é carregada

## Passos para reprodução

1. Logar em ambiente de homologação ou produção
2. Clique em 'Configuração no menu'

### Comportamento atual

Página não é carregada

### Comportamento esperado

A página ser carregada

## Contextos, prints e informações adicionais

[image]
 
Aparece isso aqui no log

## Possível Solução

Caso você tenha algum sugestão de onde começar a procurar o problema ou uma
solução para ele escreva aqui.

tmcstiago, 2019-06-11:
Este erro acontece devido a campo "nome" estar nulo.

Para resolução deste problema, sugiro alterar o atributo "key" no component "MenuItem" da função "itensSelection"

icaropires, 2019-06-14:
Ainda não tem nenhum assignee. Parece um bug prioridade, não?

PR #260 (2019-06-15): Resolvendo erros da página de configuração
## Problema/Funcionalidade

Foi corrigido alguns erros que estavam quebrando a página de configuração.

### Como foi implementado

* Para o erro da #241 foi realizada uma verificação no momento em que recebe a resposta de todas as calibrações carregadas no sistema;
* Para o erro da #259 está sendo realizada uma verificação no momento em que o upload do arquivo é realizado.

### Autores

@tmcstiago 

### Issues

resolves #241 
resolves #259
closed 2019-06-15 by tmcstiago